AstoCAD/FreeCAD #1: Damaged MacOS App - Build 40208 Silicon ARM64

MDsza, 2025-02-14:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Downloaded dmg-File can not be opened. Error Message "You cannot open the program “AstoCAD.app” because it may be damaged or incomplete."

### Full version info

```shell
AstoCAD_weekly-builds-40208-macOS-apple-silicon-arm64
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

DerFuchs, 2025-02-14:
Same here, MacBook Pro M1, running latest MacOS

bennetto, 2025-02-15:
Same here, MacBook Pro M1, MacOs 14.1.1 (23B81).

ttomasini, 2025-02-15:
Same here, MacBook Pro amd64, macOS Version 15.3 (24D60)

```
AstoCAD_weekly-builds-40208-macOS-intel-x86_64
```

MDsza, 2025-02-16:
Are there any **updates** on when the working Mac versions of AstoCAD will be available for download? On the membership website, the “Downloads” section has shown “No downloads available yet” for several days now.

If this were a purely open-source FreeCAD project, I wouldn’t ask. However, since AstoCAD is a planned financially supported project, more transparency and communication regarding availability would be welcome. So far, there has been little information or an opportunity to test the software.

I truly appreciate your huge efforts since years simplifying FreeCAD so much, as reflected in the positive response to your first AstoCAD video on YouTube. I hope this initial hurdle is just a temporary challenge and that things will work out well for you, too. Looking forward to further updates with kind regards from Switzerland.

vlbosch, 2025-02-17:
Some other apps like SuperSlicer are also affected by this issue. It can probably be solved by running: xattr -cr /Applications/AstoCAD.app

MDsza, 2025-02-17:
@vlbosch: thanks for the hint, but it doesn't work for me, too.

tostr7191, 2025-02-17:
I get the same problem, and tried `xattr -c /Applications/AstoCAD.app/` to no avail. However, the following works for me:

```
cd /Applications/AstoCAD.app/Contents/Resources
./AstoCAD
```

Starts as I would expect. So I would assume there is a problem with the structure of the dot app, not the binary itself. 

Also, can I just rave that this is the first time any FreeCAD* version on my HiDPI screen on a mac was just usable & readable? That is so very nice!!

tostr7191, 2025-02-17:
Just for anyone looking to fix the issue, this is the output of running the command in a terminal. Looks completely fine to be, but then again, I am not a FreeCAD dev.

```
Error: Failed to open library "/Library/Frameworks/3DconnexionNavlib.framework/3DconnexionNavlib"! Error: dlopen(/Library/Frameworks/3DconnexionNavlib.framework/3DconnexionNavlib, 0x0005): tried: '/Library/Frameworks/3DconnexionNavlib.framework/3DconnexionNavlib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/Library/Frameworks/3DconnexionNavlib.framework/3DconnexionNavlib' (no such file), '/Library/Frameworks/3DconnexionNavlib.framework/3DconnexionNavlib' (no such file), '/System/Library/Frameworks/3DconnexionNavlib.framework/3DconnexionNavlib' (no such file, not in dyld cache)!
AstoCAD 1.1, Libs: 1.1.0devR40208 (Git)
© FreeCAD team and AstoCAD, 2001-2025
AstoCAD is based on the FreeCAD project and released under the terms of the LGPL2.1+ License.
AstoCAD wouldn't be possible without the FreeCAD community.

          d8888          888             .d8888b.        d8888 8888888b.
         d88888          888            d88P  Y88b      d88888 888  "Y88b
        d88P888          888            888    888     d88P888 888    888
       d88P 888 .d8888b  888888 .d88b.  888           d88P 888 888    888
      d88P  888 88K      888   d88""88b 888          d88P  888 888    888
     d88P   888 "Y8888b. 888   888  888 888    888  d88P   888 888    888
    d8888888888      X88 Y88b. Y88..88P Y88b  d88P d8888888888 888  .d88P
   d88P     888  88888P'  "Y888 "Y88P"   "Y8888P" d88P     888 8888888P"

(qt.qpa.fonts) Populating font family aliases took 234 ms. Replace uses of missing font family "Courier" with one that exists to avoid this cost. 
      Migrating Start Workbench to Start command... done.
2025-02-17 20:49:10.798 AstoCAD[27136:2269990] +[IMKClient subclass]: chose IMKClient_Modern
2025-02-17 20:49:10.798 AstoCAD[27136:2269990] +[IMKInputSession subclass]: chose IMKInputSession_Modern
```

MDsza, 2025-02-17:
Thanks @tostr7191 for you input - very appreciated... I tested your **AstoCAD** **binary** hint on my macOS 15.3.1 (24D70), but I couldn’t get it to start. Here are my findings so far:
	•	The .app structure exists, but the folder /Applications/AstoCAD.app/Contents/MacOS is empty.
	•	There is no executable file (AstoCAD or similar).
	•	Running xattr -cr and sudo xattr -rd com.apple.quarantine does not change anything.
	•	Trying to launch from /Applications/AstoCAD.app/Contents/Resources also fails because no executable file exists.
	•	The suggestion to install 3dconnexion (which was mentioned in your previous comment .. for SpaceMouse-Support?) does not work because no such package exists in Homebrew.

**Conclusion**: The app seems to be incomplete or incorrectly packaged, as the main executable is missing. A new, complete build would be appreciated.

Thanks for your efforts! I hope this issue can be resolved soon.

MDsza, 2025-02-17:
**Good news!**

The workaround mentioned by @tostr7191 has now also worked for me by following these steps:
	1.	Open Finder → Applications → Right-click on AstoCAD and select Show Package Contents → Navigate to:
                Applications/AstoCAD.app/Contents/Resources/bin
	2.	Double-click AstoCAD (macOS security features will block the app the first time, as its security cannot be verified).
	3.	Open System Settings → Privacy & Security.
	4.	Scroll down to the section “App was blocked”.
	5.	Click “Open Anyway” or “Allow”.
	6.	Repeat steps 1 and 2.
	7.	A terminal window will open… AstoCAD will start.

Until this issue is resolved, steps 6 and 7 must be repeated each time you open AstoCAD.

[image]

tostr7191, 2025-02-18:
@MDsza I made a typo in my original message, you need to `cd /Applications/AstoCAD.app/Contents/Resources/bin`, then you can `./AstoCAD`. If you are not very familiar with the terminal (you will have it as `Terminal.app` on your mac), you can run the binary directly without `cd` like this: `/Applications/AstoCAD.app/Contents/Resources/bin/AstoCAD`.

dan-and, 2025-02-22:
> **Good news!**
>
> The workaround mentioned by [@tostr7191](https://github.com/tostr7191) has now also worked for me by following these steps: 1. Open Finder → Applications → Right-click on AstoCAD and [...]

It is also required (At least in 15.3.1), that xattr -cr /Applications/AstoCAD.app is done as well. So changing the xattr -cr is mandatory as step 0 ;-)

cbenhagen, 2025-02-22:
Instead of running it from the command line you can also link the binary to the correct place. So the full fix would be:

```bash
xattr -cr /Applications/AstoCAD.app
ln -s /Applications/AstoCAD.app/Contents/Resources/bin/AstoCAD /Applications/AstoCAD.app/Contents/MacOS/AstoCAD
```

From now on you can run the app as you would expect.

PaddleStroke, 2025-02-24:
Thank you guys for looking into it, and sorry that I could not check this out sooner as I was away with a broken laptop.

Today I have worked on fixing the woocommerce/paypal as there were annoying bugs between these 2. But now my priority is to fix the Mac build and fix the custom title bar for mac and linux.

Btw as Mac users, if you could give me some hints on what the custom title bar should be doing but is not doing yet? If you have some videos to show me how a window should behave it would be great. For instance on windows click and drag moves it, click an edge resize, double click maximize, dragging to the top maximize ...
Thanks

cbenhagen, 2025-02-25:
The first part of this issue is fixed thanks!

Signing and notarization can be done in the CI/CD pipeline so no additional hardware should be needed for that. But you definitely need an Apple Developer Program membership which costs 99 USD / year. If you’re a nonprofit organization, accredited educational institution, or government entity and meet certain requirements, you can request an Apple Developer Program [fee waiver](https://developer.apple.com/support/fee-waiver/).

PaddleStroke, 2025-02-25:
Great thanks for the confirmation. Then perhaps it's best if I close this issue and open a new one specifically for the signing.
99/y is as bad as I thought, I will need to check this too.

closed 2025-02-25 by PaddleStroke

PaddleStroke, 2025-02-27:
I have some people telling me the 40451 ARM build is still not working. So I wonder perhaps only intel is fixed.
@cbenhagen you confirmed it's ok, on intel or arm?
@MDsza you're on ARM right? Can you try the 40451 build and let me know if the problem 1) is fixed?

cbenhagen, 2025-02-27:
I just re-tried 40451 on arm on macOS 15.3.1. This is the error users get:

[image]

This can not be fixed by going to System Settings → Privacy & Security. The only workaround I know of is to run:
`xattr -cr /Applications/AstoCAD.app` which clears all attributes recursively.

PaddleStroke, 2025-02-27:
@cbenhagen can you please tell me, was it the same ARM computer that you used before when you confirmed 1) was solved?
If yes, what file version was it?
If no, then it confirms the ARM version is failing but not the intel

MDsza, 2025-02-27:
Hi PaddleStroke, great to have you here again (with a running notebook again :-) ). 
Yes I am on ARM (MacBook Pro M4) Just tested the 40451 ARM build (webpage is even quite slow ...since the beginning): 

**App-Icon**: like at the last version, the App Icon has still a grey "forbidden" sign, after having extracted the App from the dmg-Downloadfile.
[image]

Same result like @cbenhagen  mentioned just befor: after having run `xattr -cr /Applications/AstoCAD.app` in the terminal AstoCAD runs fine...

Hope that helps...

PaddleStroke, 2025-02-27:
I think this error is maybe also because macOS Gatekeeper is blocking the unsigned app.
I will try to setup the signing of the app asap. I'm just not sure how the FreeCAD CI is doing it. I see there is a macos_sign_and_notarize.zsh file, but I don't see how it is triggered or how to set it up.

PaddleStroke, 2025-02-27:
Oh FreeCAD has the same issue indeed. It's there https://github.com/FreeCAD/FreeCAD-Bundle/issues/291
I was looking at the differences between astocad and freecad, so of course I wouldn't find it :)