AstoCAD/FreeCAD #3: [Feature Request] Option to bring back Close Sketch button to the Task Panel

Footz355, 2025-02-15:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Please just hear me out. The idea was to de-clutter the Task Panel. But it caused an unnecessary in my opinion running with a cursor through the whole screen to close the sketch. Mind you, I have moved the Close Sketch button in the task button all the way to the upper right-hand side to be close to the Task Panel to mitigate looking for the Close Sketch button with my mouse.

[image]

But there is another issue here. Every feature, like Pad, Pocket etc, still keeps it's OK button (close feature edit button) in the Task Button. Do you intend to remove those buttons as well and replace them with one general "APPLY" main menu bar button for the sake of consistency?  Or will we have click somewhere else to close the sketch, and somewhere else to apply a feature?

I hope I made my point clear. 
Personally for me there was nothing wrong with the OK button located at the Task Panel but I'd like to hear other users opinions
Thanks.

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.40208 (Git) Conda AppImage
Build type: Release
Branch: AstoCAD
Hash: 760cbe471b1cf8d7408e7cbf48f876960012a59c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/Dark/Fusion
Logical/physical DPI: 107.48/108.888
Installed mods: 
  * sheetmetal 0.7.11
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

NullByte28, 2025-02-18:
Hello
We have to get used to it, there's nothing wrong with that. Users of proprietary software will also have to get used to the changeover to FreeCAD.
With large and complex sketches, scrolling to the close button is annoying, the new button in the Sketcher editing mode panel is a pleasant help. However, other functions such as Pad, Poket, Fillet, Chamfer etc. should also be terminated by this button.

Translated with www.DeepL.com/Translator (free version)

Footz355, 2025-02-20:
> Hello We have to get used to it, there's nothing wrong with that. Users of proprietary software will also have to get used to the changeover to FreeCAD. With large and complex sketches, scrolling to [...]

If scrolling to the close button is annoying, than what do you call when you have to swing your mouse all the way to the opposing edge of the screen. The issue we are discussing here is highly subjective, I know, but if the OK/Close button will remain independent, maybe there would be an option for it to be placed in an independent section of the task panel, above the scrollable data input section. Just my 2 c.

PaddleStroke, 2025-02-20:
I can understand that for those who are used to the close button, the change can be annoying. So yes I think adding a parameter is totally acceptable. I will add it for next release.

Footz355, 2025-02-20:
> I can understand that for those who are used to the close button, the change can be annoying. So yes I think adding a parameter is totally acceptable. I will add it for next release.

I don't want to sound annoyed really. It's a bit of a shock that's all. We need bold moves. Just like I said in original post, if you have a "road map" for a 'unified' APPLY/CANCEL button independent completely from the sketcher and other feature's TASK PANEL , maybe that's the way to go.

PaddleStroke, 2025-02-25:
For me the situation is reverse, I usually had to travel the mouse all the way to the right, when my mouse is usually closer to the toolbars. So the toolbutton is more practical on this regard for me. Also I don't really considered the sketch editing as a 'command' as a pad operation is. Which is why I prefered to remove it.
Anyway a parameter to add the button back is small effort so I can definitely at least add a parameter.

The change is added to my local branch and will be in the next release. I also made a [pull request to FreeCAD main](https://github.com/FreeCAD/FreeCAD/pull/19849)

Note the PR to freecad keeps the close button activated by default.

When next release of AstoCAD comes out, you will be able to bring back the close button by adding a parameter Preferences/Mod/Sketcher/ShowCloseButton and set it to true.

PaddleStroke, 2025-02-25:
I don't know yet about adding a Ok/cancel button to the toolbr for all operations. I'll think on it.

closed 2025-02-25 by PaddleStroke

PaddleStroke, 2025-04-03:
After some more thinking, I think I changed my mind on this question. Many users seems to dislike this change. Besides discussing it with the DWG (freecad's design working group) it appears that I was the only one prefering removing the close button.
Then with further discussion, I realized that what was annoying me, was that the close button was kind of floating there above the 3 widgets.

So my new proposal is the following : 

[image]

- Remove the OK/Cancel from the toolbar.
- Remove the grid tool from the toolbar.
- Rename the 'solver message' widget 'Sketch edit'. This widget has a setting button holding the grid, snap and rendering order all in the same place.
- The OK/ cancel buttons are modified so that they do not float around. Instead they are next to the "Sketch Edit" widget.
- The OK/Cancel buttons look like this for all the tasks.

Thoughts ?