AstoCAD/FreeCAD #4: The dragbar functionality does not work in Linux (xfce)

Hoglet, 2025-02-15:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Only way to move the window is with keyboard "modifier" (<Alt> in xfce)

### Full version info

```shell
Having an option in preferences that allows the window manager draw the window might be a stop gap.
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Hoglet, 2025-02-17:
While clicking around it suddenly started to use the window manager so I can now move. I have no clue where that setting is but it also made that the menu is constantly visible. Just a FYI in case somebody else get the same problem.

PaddleStroke, 2025-02-26:
Basic functionalities:
- click and drag the title bar should move the window
- double click the title bar should maximize/minimize.

Then there is the advanced functionalities which I am not sure how to handle. For instance I have a ubuntu laptop on which I tested, if I drag the window to the top it maximize, if I drag to one side it takes half of the screen... But for those I think it's maybe different for each linux distro no?

PaddleStroke, 2025-02-27:
New builds are up : AstoCAD_40464
It is now possible to move the dialog by click and drag on the titlebar. And also maximize/restore by double click on the title bar.
I have tested on ubuntu. @Hoglet if you can confirm it is also working on your side?

I'll close this issue now. If it's not working properly still on xfce feel free to reopen it

closed 2025-02-27 by PaddleStroke