AstoCAD/FreeCAD #5: [Assembly] [Feature request]: fixed joint with precalculated offsed

kwahoo2, 2025-02-15:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Allow fixing part to part without moving it's position - an precomputed offset has to be applied.
Described here https://forum.freecad.org/viewtopic.php?t=93145 

I made some very basic implementation:
https://www.youtube.com/watch?v=TTSxNNWtb2U

https://github.com/kwahoo2/FreeCAD/commit/338bc5dee2c7507689939df57c7b46cbd36005f4

At the end it should behave like a GroundedJoint joint, but in relation to another part - if you move a part in the joint edit mode, the new offset should be calculated.

### Full version info

```shell
(not able to download AstoCAD Linux yet)
OS: Fedora Linux 41 (KDE Plasma) (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * OpticsWorkbench 1.0.26
  * fasteners 0.5.35
  * sheetmetal 0.6.13
  * Curves 0.6.53
  * freecad.gears 1.3.0
  * freecad-xr-workbench 0.9.4
  * kicadStepUpMod 11.4.4
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

PaddleStroke, 2025-02-20:
This sounds like a reasonable feature request. This perhaps belong to the FreeCAD's repository though.

PaddleStroke, 2025-02-20:
Yes it can be only in AstoCAD for some time without problem. I just meant that the feature request is also valid for FreeCAD.

PaddleStroke, 2025-02-28:
i just seen the video, your implementation looks good. Is there something missing in it? If you share the code I can use it if you want.

kwahoo2, 2025-02-28:
>If you share the code I can use it if you want.

The code is linked under the video. Please use it if it can be useful.
The missing part is the joint redefinition. An user should be able to recalculate the offset by simply translating one object relative to another, in similar fashion as GroundedJoint can be redefined.

kwahoo2, 2025-03-04:
I'd like to avoid adding the Preserve Position property. It was my initial idea to make offset update with the dragger possible.

If there would be a  button in the taskbox, it should be also used to forbid relative part movement during joint creation. So. 3 main states:

- Preserve relative offset (active before a joint creation)
- Change initial offset (a joint is created already)
- Confirm initial offset

As for other types of joints, it will be tricky, probably. But it would be useful for gears, as an user should have a way to adjust the gears meshing.

PaddleStroke, 2025-04-18:
Build 41331 posted include this feature.

closed 2025-04-18 by PaddleStroke