AstoCAD/FreeCAD #6: Would love to see a Changelog showing the different features you've added

luzpaz, 2025-02-15:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Would you consider that? 

### Full version info

```shell
n/a
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

PaddleStroke, 2025-02-20:
Yes I need to have a detailed log file indeed

PaddleStroke, 2025-03-03:
So after thinking about it, there's already the feature list on the website. And making a more detailed list will be annoying to maintain. Also there is the commit history that already gives a clearer view of the changes.

closed 2025-03-03 by PaddleStroke

luzpaz, 2025-03-03:
Maybe consider auto-generating from the commit history?