AstoCAD/FreeCAD #10: AppImage failes with `execv error: No such file or directory`

davidheiss, 2025-02-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Can't open `AstoCAD_weekly-builds-40208-Linux-x86_64.AppImage `
Error message is `execv error: No such file or directory`

I also get this error when trieing to extract the files.
```
./AstoCAD_weekly-builds-40208-Linux-x86_64.AppImage --appimage-extract
This doesn't look like a squashfs image.
Failed to open squashfs image
```

### Full version info

```shell
Can't
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

PaddleStroke, 2025-02-25:
Can you please check the size of the appimage ? Perhaps there was a problem during download and the archive is incomplete/corrupted?

CIKoolK, 2025-02-26:
I will report that the b40208 .AppImage is working fine, both intact and extracted on my platform--_Mint v22.1/MATÉ:_

PaddleStroke, 2025-02-27:
@davidheiss @MarekGajdosik can you guys please confirm if you can open the weekly builds of FreeCAD correctly?

davidheiss, 2025-02-27:
For me, the error persists.

```
./AstoCAD_40451-Linux_34b885f2ac912cfc23ed339bd0f33484.AppImage --appimage-extract
This doesn't look like a squashfs image.
Failed to open squashfs image
```

PaddleStroke, 2025-02-27:
I meant the weekly-build of FreeCAD, as opposed to to AstoCAD.

davidheiss, 2025-02-27:
I think there is an ISSUE with AppImageLauncher (https://github.com/TheAssassin/AppImageLauncher/issues/599) , after uninstalling it and disabling it in binfmt_misc both work.

PaddleStroke, 2025-02-27:
Oh good. Then this can be closed ?

CIKoolK, 2025-02-27:
_b40408_ is working as intended here: _Mint v22.1/MATÉ_...

closed 2025-02-27 by davidheiss

MarekGajdosik, 2025-03-01:
Oh my, you are correct, Issue was caused by AppImageLauncher.

CIKoolK, 2025-03-01:
Sometime late last year _AppImage_ switched to using a new compression algorithm that _AppImageLauncher_ could not deal with (it appears _AppImageLauncher_ was last updated 4 years ago), I've never used it but have been extracting _.AppImag_e bundles to individual folders for years--it make the application load faster ant run more smoothly.

I developed a bash script to do this, revised for _FreeCAD_, and now _AstoCAD_, to now:

- Open a file dialog to select the source _.AppImage_ file;
- Extract the file to the _.AppImage_ default ._/squashfs-root_ folder;
- Rename that folder _FreeCAD-live_ (or _AstoCAD-live_)--after deleting any such existing folder;
- Optionally create a desktop launcher;

If this is something you might be interested in "_let out a whoop!_" (my Florida raised grandson's favourite directive) and I can get it to you...