AstoCAD/FreeCAD #11: Does not connect to spacenav

Hoglet, 2025-02-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When starting it does not seem to try to connect to spacenavd, excerpt from log files for for freecad and astocad respectively:
``` shell
hoglet@Big:~$ grep spacenav /tmp/freecad_1.1.log 
Log: Connected to spacenav daemon
Log: Disconnected from spacenav daemon
hoglet@Big:~$ grep spacenav /tmp/astocad.log 
Log: Couldn't disconnect from spacenav daemon
```

Astocad only tries to disconnect, not connect.

### Full version info

```shell
OS: Devuan GNU/Linux 6 (excalibur/ceres) (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.40208 (Git) Conda AppImage
Build type: Release
Branch: AstoCAD
Hash: 760cbe471b1cf8d7408e7cbf48f876960012a59c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/Light/Fusion
Logical/physical DPI: 110/109.201
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

PaddleStroke, 2025-03-04:
@Hoglet a few questions : 
- Have you tried in the past the builds from Ondsel, and if so was it working with Ondsel builds?
- Do you know where the connection happens in the code perhaps? I have no idea where to start looking

Hoglet, 2025-03-05:
I am using the spacemouse in freecad 1.0, I have not aware of using any Ondsel builds.
I can dive into the code and see if I can find it.

Hoglet, 2025-03-05:
I suspect https://github.com/AstoCAD/FreeCAD/blob/076899205c7b6c3d3ae31e27a3144568d704f132/src/Gui/GuiApplicationNativeEventAware.cpp#L62

Maybe you are missing the 3D connexion SDK  or you manually need to set the _USE_3DCONNEXION_SDK in makefile?

Hoglet, 2025-03-05:
Can this be it:

https://github.com/FreeCAD/issues-backup/blob/82e54c0cee590d8f52a56e91b2baf59582c75a16/issue-attachments/files/CMakeCache.txt#L435

PaddleStroke, 2025-03-06:
Can you please check in current 1.1 dev version if spacemouse works correctly?

I have not changed anything regarding spacemouse in the AstoCAD/FreeCAD repository. So the problem is unlikely to be there.
It is more likely in the 'Bundle' or the 'Feedstock' repositories, which are the scripts that are generating the AstoCAD builds.

Having said that something comes to mind actually. AstoCAD is not using the same 'user' folder, ie the folder where the user.cfg is located. On windows that is : 
C:\Users\Sky_l\AppData\Roaming\FreeCAD

And AstoCAD is using : 
C:\Users\Sky_l\AppData\Roaming\AstoCAD

As to avoid messing with user's FreeCAD's preferences.

So I am thinking perhaps 3D_Connexion is storing or trying to access things in the Roaming\FreeCAD.

PaddleStroke, 2025-03-06:
@Hoglet can you try removing temporarily the 'Branding.xml' from the image bin folder? This should put back the 'user' folder to "Roaming\FreeCAD" equivalent on linux. Then try to launch astocad and see if the problem is there still.