AstoCAD/FreeCAD #14: Threat report in AstoCad latest weekly build (copied from FreeCAD)

bitbasher, 2025-02-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

this is a second report on this issue .. [first report ](https://github.com/FreeCAD/FreeCAD/security/advisories/GHSA-98fx-j9r2-956f)was closed because after some work with the FreeCAD version 1.1 installation that the problem WAS NOT in the FreeCAD part of the AstoCAD installation.. i thought it was my fault for using 7Zip incorrectly to unpack the AstoCAD zip file.

I have now downloaded and re-installed AstoCAD and it ran just fine. The first step in getting the app ready is to install all of the extra workbenches and macros that i use .. I got part way thru all that when my anti-virus scanner detected a Trojan in PathApp.pyd again. And this time i read the file path correctly as /AstoCAD/lib/PathApp.pyd so it is now clear that the threat is only in there, and NOT in the FreeCAD 1.0 nor in FreeCAD 1.1, both of which are now installed and running on my machine

[image]

The PathApp file was moved to Quarantine so there is not much i can do in AstoCAD to test further. It still starts and gives me the UI, but i cannot do anything that looks into the file system.

I started up FreeCAD 1.1 and made a few parts to give my AV scanner time to find a threat .. but it did not so it looks like AstoCAD is the only one with the problem. I have also run FreeCAD 1.0 with no sign of a problem.

Details

I have done a binary compare of the PathApp.pyd from AstoCad and FreeCad 1.1 and have seen they are different.
I have also done a binary compare between PathApp.pyd from AstoCAD and FreeCAD 1.0 and they are different also .. but there are some large blocks of difference that might be a viral inclusion

I have run my virus scanner (AVG) on the whole disc now and only the AstoCAD package had the "infected" file .. it might be this is a false positive so .. grain of salt?
PoC

ah .. i just downloaded the packages and used the Windows right click contextual menu to select Extract All .. then ran the apps by going into the respective /bin directories and double-clicking to run the apps ..
this is my Windows info

[image]

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.40208 (Git) Conda
Build type: Release
Branch: AstoCAD
Hash: 760cbe471b1cf8d7408e7cbf48f876960012a59c
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/Belgium (en_BE)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/Light/Fusion
Logical/physical DPI: 96/91.7938
Installed mods: 
  * 3D_Printing_Tools
  * addFC 1.3.0
  * boltsfc 2022.11.5
  * btl 0.9.9
  * CurvedShapes 1.0.13
  * Curves 0.6.58
  * Defeaturing 1.2.3
  * DynamicData 2.71.0
  * fasteners 0.5.37
  * freecad.gears 1.3.0
  * FreeGrid 2.2.0
  * Gridfinity 0.9.4
  * sheetmetal 0.7.11
  * woodworking 0.21.2.33771.0.23.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

bitbasher, 2025-02-18:
i reported this as a security issue in what i thought was the AstoCAD repository but it created the report in FreeCAD .. chennes said they could not transfer it to your side so i have copied it here

PaddleStroke, 2025-02-20:
Hello,
Sorry about this issue. I am not sure what this is about as I don't remember changing the PathApp.pyd file. I need to check what this file is and what is the difference between astocad's build workflow and freecad's as I guess this is some sort of library.

PaddleStroke, 2025-02-25:
Ok so apparantly PathApp.pyd is a compile file generated by the CAM workbench. I have not touched anything in CAM so I'm not sure what to think.

My current hypothesis is that this happened only at the point where AstoCAD build. ie it was not happening in 1.0, and is not happening in latest dev version. But maybe it happens at the point where AstoCAD is being build.
Long story short I have just rebased and will push and trigger the builds, then we'll see if the new builds still have this problem.