AstoCAD/FreeCAD #15: DLL missing and DLL errors from Windows Membership version

cdashdashb, 2025-02-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I'm getting a host of errors upon starting AstoCAD.exe, first one is:

"
AstoCAD,exe - Bad Image
C:\Folderstructure_on_desktop\FreeCADGui.dll is either not designed to run
on Windows or it contains an error. Try installing the program
again using the original installation media or contact your
system administrator or the software vendor for support. Error
status 0xc000007b.
x
"

Second error:
"
AstoCAD.exe - System Error
The code execution cannot proceed because
Qt5Gui_conda,dll was not found. Reinstalling the program
may fix this problem.
"

"
Same as above for QTCore_conda.dll
"

"
same as above for Qt5Widgets_conda.dll
"

I've seen no mention of anyone else having this issue, also I've seen no mention of additional requirements for install? Any insight?

### Full version info

```shell
AstoCAD_weekly-builds-40208-Windows-x86_64
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

PaddleStroke, 2025-02-20:
I am not sure where this issue could come from yet.
Can you please let me know if you can succesfully use the FreeCAD weekly builds? So that I know if the problem is specific to astocad or not.
Thanks

cdashdashb, 2025-02-21:
I tried out the most recent FreeCAD weekly (40303), and it works. I suspect that a potential issue could be the zip archive? I gave unzipping the archive another go and there are some errors when decompressing. Also the download needed to be resumed multiple times, so potentially there could be corruption there?

wireedm1, 2025-02-21:
Same issue here.  Became a member, downloaded AstoCAD_weekly-builds-40208-Windows-x86_64, and it gives me a LOT of errors when extracting.

PaddleStroke, 2025-02-25:
This definitely sounds like a corrupted download issue.
I am still working on solving the download issue.

wireedm1, 2025-03-03:
This worked, thank you!

I opened AstoCAD 1.1.0dev from the download and I'm impressed.  As a Solidworks user everyday, AstoCAD is similar enough to be intuitive to use.

Thank you for your work on it!

closed 2025-03-03 by PaddleStroke