AstoCAD/FreeCAD #16: AstoCAD crashes on start when the language is German and the Custom tile bar is disabled

tak-21, 2025-02-21:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Short version: switch to German language, disable the Custom tile bar, restart AstoCAD --> crash

As the custom tile bar is currently not working in linux, I disabled it. When I wanted to open AstoCAD again, it crashed / did not start. I had to delete all local files in my home dir regarding AstoCAD before it would start again.
After some trial and error I realized, that if I keep English as program language, this is not happening. As soon as I switch to German again it crashes.

### Full version info

```shell
v 1.1.0dev; 40208 /Git); Linux Mint 22.1, HASH 760cbe4
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

PaddleStroke, 2025-02-24:
Thanks for the report. I will look into this, I'm not sure yet what can cause this

PaddleStroke, 2025-02-28:
I can reproduce, if frameless is deactivated, changing to any other language crashes.

PaddleStroke, 2025-03-03:
Fixed it. I'll let you know when I push the new builds including this fix.

PaddleStroke, 2025-03-03:
Build 40515 is up on the website and fix this.

closed 2025-03-03 by PaddleStroke