AstoCAD/FreeCAD #17: [Sketcher] [macOS] HiDPI scaling for fonts and symbols

cbenhagen, 2025-02-22:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In Sketcher (and maybe other workbenches) fonts and symbols are not scaled correctly on macOS HiDPI screens. As HiDPI (Retina) screens have been the default on Apple hardware for a very long time so I would expect most macOS users to be affected by this.

[image]

### Full version info

```shell
OS: macOS 15.3.1
Architecture: arm64
Version: 1.1.0dev.40208 (Git) Conda
Build type: Release
Branch: AstoCAD
Hash: 760cbe471b1cf8d7408e7cbf48f876960012a59c
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/Dark/Fusion
Logical/physical DPI: 72/151.5
```

### Subproject(s) affected?

Sketcher

### Anything else?

Upstream issue linked for premium membership prioritization: https://github.com/FreeCAD/FreeCAD/issues/10982

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

cbenhagen, 2025-02-23:
Measurements have the same issue:

[image]

PaddleStroke, 2025-02-24:
I'll check this out as soon as I can, however I first have to fix the mac and linux builds and their custom title bars as this is quite critical. Once things are rolling smoothly I'll try to fix this

PaddleStroke, 2025-03-04:
I think I found the issue. I'll let you know when the new build is up for you to test

PaddleStroke, 2025-03-05:
Buikd 40544 is up on the website. If you can give it a go and confirm if it's good now? Note this was only for sketcher not dimensions

PaddleStroke, 2025-03-05:
Erratum! I forgot to push after making this commit so it is not in 40544! Sorry I'll push new builds later

PaddleStroke, 2025-03-06:
@cbenhagen I have just pushed new builds today, 40545, in which this bug should be fixed. Can you please test and let me know?

PaddleStroke, 2025-03-06:
Great. I'll close this then.
For the symmetry icons, I guess for some reason they must not be using the sketcher's font size like the rest of the icons.

But they appears to me to be reasonably sized as they are don't you think?

closed 2025-03-06 by PaddleStroke