AstoCAD/FreeCAD #19: AstoCAD Window Remains White After Maximization

toskium, 2025-02-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

### Issue Description:
When AstoCAD is minimized in the following state:

[image]

And then maximized by left-clicking its icon on the taskbar:

[image]

The main window remains completely white and does not display any content.
### Workaround:

Double-clicking anywhere in the menu bar region forces the main window to appear, but it does not restore in maximized mode:

[image]

After this, the window can be manually maximized again.

### Expected Behavior:
When restoring AstoCAD from the taskbar, the main window should properly display its contents in the previously maximized state.

### Steps to Reproduce:
- Minimize AstoCAD.
- Restore it by left-clicking the taskbar icon.
- Observe that the window remains white.
- Double-click the menu bar region as a workaround.
- Manually maximize the window again.

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40208 (Git) Conda
Build type: Release
Branch: AstoCAD
Hash: 760cbe471b1cf8d7408e7cbf48f876960012a59c
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/Germany (en_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/Dark/Fusion
Logical/physical DPI: 96/92.6073
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

PaddleStroke, 2025-02-24:
Hmm I cannot reproduce this. I wonder if this is a windows 11 bug? I'm using windows 10 on my end.
Anyone else with windows 11 can confirm this?

toskium, 2025-02-25:
@PaddleStroke is there anything I can do to help pinpointing the issue? Maybe some sort of debug logging?

PaddleStroke, 2025-02-25:
@toskium if you have several windows machines available, if you could test on the different ones that would perhaps highlight what is triggering this. I'm thinking either a windows version or a hardware thing like the GPU model.
Does it happen 100% time when you minimize then restore? Only if you restore by clicking on the windows bar or also if you use alt+tab?

toskium, 2025-02-25:
So these are the machines I have access to at the moment:

Problem **does not** occur on Windows 11 22H2 (virtual machine)
GPU: Virtual GPU, VMware Workstation, no 3D acceleration
OS: Windows 11 build 22621
Architecture: x86_64
Version: 1.1.0dev.40208 (Git) Conda
Build type: Release
Branch: AstoCAD
Hash: 760cbe471b1cf8d7408e7cbf48f876960012a59c
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/Germany (en_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/Dark/Fusion
Logical/physical DPI: 96/96.0353

Problem **does occur** on Windows 11 23H2 (physical machine)
GPU: Nvidia rtx 4080 Super
Driver Version: 566.36 (Dec. 05. 2024)
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.40208 (Git) Conda
Build type: Release
Branch: AstoCAD
Hash: 760cbe471b1cf8d7408e7cbf48f876960012a59c
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/Germany (en_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/Dark/Fusion
Logical/physical DPI: 96/108.888

Problem **does occur** on Windows 11 24H2 (physical machine)
GPU: Nvidia rtx 4090
Driver Version: 572.47 (Feb. 20, 2025)
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40208 (Git) Conda
Build type: Release
Branch: AstoCAD
Hash: 760cbe471b1cf8d7408e7cbf48f876960012a59c
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/Germany (en_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/Dark/Fusion
Logical/physical DPI: 96/92.6073

toskium, 2025-02-26:
@PaddleStroke, am I correct in assuming that the settings shown in the screenshot rule out any potential Nvidia driver issue?
If that's the case, it appears that Windows 11 23H2 (and later) might be causing the problem.

[image]

PaddleStroke, 2025-02-27:
I will try this on my wife's laptop as she's on windows 11.

PaddleStroke, 2025-02-27:
I have just tried on a windows 11, 24h2 laptop and cannot replicated this issue. Could you please make a video showing the issue?
Thanks

toskium, 2025-02-27:
I’ve just discovered that the issue only occurs on my multi-monitor setup, and specifically on the 'primary screen.' To give you a better understanding of what you're seeing in the video: I have three monitors, with the primary screen being the one visible first in the video. This is the screen where the white-app issue appears. You’ll also see me moving the AutoCAD window to my left monitor, where the issue does not occur.

https://github.com/user-attachments/assets/5c59dc3b-5db8-49ac-80d2-27c37c785f8b

PaddleStroke, 2025-02-27:
Thanks for the clarification. And so if you disconnected the additional monitors, this bug does not happen right?
I will try to replicate on my side

PaddleStroke, 2025-02-28:
Ok great, so I will be able to run some tests soon.

PaddleStroke, 2025-03-03:
Can you please advise if this happens only if the window is maximized before you minimize ? What if the application is not full screen?

PaddleStroke, 2025-03-03:
From your video I think it does not happen when the application is not full screen. If you can just confirm to make sure.

toskium, 2025-03-03:
The white-application issue only appears when the application was already maximized on the primary screen.
If I am minimizing a non-maximized application to the taskbar, the issue does not occur.

PaddleStroke, 2025-03-03:
Build 40515 is up on the website if you want to give it a test. I'm not sure it's going to fix it though.
I have another build underway that will be up tomorrow and I think that one is more likely to fix the bug. Though the first is still good to test if possible.

What is annoying is that I cannot replicated at home, because my windows laptop screen looks dead (it's black and won't turn on because of some intel iris/bios version issue that I can't fix). So I'm using my laptop only with my external monitor. So yes I can't test this primary screen issue since primary screen is out of order :(

PaddleStroke, 2025-03-04:
I have posted build 40516 if you could please test it?
Thanks

toskium, 2025-03-04:
The white screen issue is now fixed, but it seems the fix is drawing the window slightly larger than the screen resolution. It looks like the main window is cropped.

As shown on the screenshot in 40516

[image]

vs. 40515

[image]

PaddleStroke, 2025-03-04:
Ok good at least we know where it comes from.
So the thing that was causing the white screen issue, is the thing that was fixing the cropping you are seeing.
So I need to find a new fix that does not trigger the white screen.

PaddleStroke, 2025-03-06:
@toskium I have pushed builds 40545 today. Can you please check if it is solving the cropping issue and the white screen issue at the same time?

toskium, 2025-03-06:
Thank you! It works perfectly! :-)

closed 2025-03-06 by toskium