AstoCAD/FreeCAD #20: [macOS] Native title-, tool- and menubar

cbenhagen, 2025-02-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Modern macOS apps often use a combined title and toolbar ([NSWindow.ToolbarStyle.unified](https://developer.apple.com/documentation/appkit/nswindow/toolbarstyle-swift.enum/unified) // [unifiedTitleAndToolBarOnMac](https://doc.qt.io/qt-6/qmainwindow.html#unifiedTitleAndToolBarOnMac-prop)):

[image]

It would be great if AstoCAD / FreeCAD would use that.

On macOS the top of the screen is reserved for the menu bar of the currently active app. So unlike on Linux and Windows it is counterproductive to hide the menu bar under the app icon. If in full screen mode, the menu bar can auto hide on macOS.

This document explains how the title bar and its buttons behave on macOS:
https://support.apple.com/en-gw/guide/mac-help/mchlp2469/mac

### Full version info

```shell
--
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

PaddleStroke, 2025-02-27:
Ok I need to find a mac computer to develop this. Else I'm just poking in the dark.

But from my reading, I think `unifiedTitleAndToolBarOnMac` is unlikely to be what we want. Because this will likely merge the titlebar with the whole toolbar area. So it would be like you have all the toolbars in the titlebar. Which is going to look bad.

OfficialKris, 2025-02-27:
I do not think that is the case. It is very similar to the unified title bar for windows just with the window operators on the left rather than the right.

Also, a M1 MacBook Air is a good choice for a tested.

PaddleStroke, 2025-02-27:
I could just change the window operator to put them on the left and change their style. However the problem is that current custom title bar is replacing the menu by a widget. Which we don't want on mac because we don't want to touch the menu or mac can't handle it.
I have family with a macbook so I'll grab that. I just need some time setting up the IDE then making it happen.

PaddleStroke, 2025-02-27:
To be noted, I just pushed new builds : 40464
This build do not have native support for frameless for MacOS. However it is now possible to move the dialog by click and drag on the titlebar. And also maximize/restore by double click.

So this is not perfect, but at least the title bar is somehow usable now.

PaddleStroke, 2025-03-28:
I confirm that `unifiedTitleAndToolBarOnMac()` does not work properly.
I will investigate further possibilities.

PaddleStroke, 2025-12-11:
still relevant