AstoCAD/FreeCAD #22: AstoCAD Sketcher toolbar

jomanxok, 2025-02-27:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Lost sketcher tool icons in tool bar, sketcher menu tool bar is working. 

### Full version info

```shell
AstoCAD 1.1.0dev
```

### Subproject(s) affected?

None

### Anything else?

I was toggle viewing a TechDraw sheet while editing a sketch in a different model file.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

PaddleStroke, 2025-02-28:
You mean the sketcher toolbar disappeared after switching to a techdraw page right? If you can provide precise reproduction steps it would be helpful.
I think this is a problem in upstream FreeCAD

jomanxok, 2025-02-28:
I was in a different file from the Techdraw sheet using Sketcher WB. The
sketcher dialog box would appear on the Techdraw screen. In this case you
also need to change WB as you toogle, really can't give the exact sequence.
I think I will need to reinstall AstoCAD to fix this to make Sketcher
usable. Will preference config backup files be a problem?

jomanxok, 2025-02-28:
BTW, I have 1.0, 1.1, and AstoCAD installed.

PaddleStroke, 2025-04-18:
Please let me know if you find reproduction steps at some point.

jomanxok, 2025-04-18:
Lets consider this closed. One other thing I see that padding cannot be toggled off in feature tree and drives width of attachment editor. Can padding be driven by text string width then and attachment editor have its own width? The attachment editor is an intermittent task and gets in the way of long multi-body tree views. Attachment editor floating and movable panel? Still would like to see toggle axis cross in sketcher, its an annoyance that seems like an easy fix.

PaddleStroke, 2025-04-22:
Sorry but I can't get it. Can you please make a video so that I can see what you mean?

PaddleStroke, 2025-04-22:
For the ' toggle axis cross in sketcher' is there an open issue in FreeCAD's tracker?

jomanxok, 2025-04-22:
Hello, l am experimenting with a panel layout that would like the tree view extend the full height of the screen and have the task panel floating and remember its location and size. I am coming from Fusion so know the font padding in the tree view and can understand that to toggle from light and dark screen but it should self correct for that. Padding seems unnecessary in dark mode.
 In the screen shot you see that the task panel does not return to full width, I can pull that wider but it would be nice if it did that automatically. It seems we work from left to right, and save mouse travel this seems good to me. In sketcher maybe the panel makes more sense on the right side of the screen, but regardless it would be nice if the panel would have that flexibility.
As for the axis cross in sketcher, why is it there? As long as the snap is preserved its not needed along with the default color scheme is very confusing. If that is a big task to change or implement a toggle off option I can live with it I guess. Fusion has no axis cross in its sketcher for example.
Just to clarify the sketcher dimension Name function is a setting that I discovered in AstoCAD (??) as it seemed to be have been enabled by default in one of your versions. seems to be very useful in a lot of cases. I do keep that enabled now. Notice how task panel location overlaps tree view by default in sketcher.
Just experimenting, hope this helpful.

[image]
  

[image]

jomanxok, 2025-04-22:
Did not mention that with both Freecad and Astocad open the space mouse function works with both. Revision 41331.

jomanxok, 2025-04-24:
A couple of issues I have found in the latest revision of Astocad. 
1.  If I have the task panel as floating the panel persist if I open the start page. It also does not always return to full width view of the panel in its last location, the panel is full width but not the dialog. What is the purpose of the persistent blue docking box and its irrational behavior?
2.  In trying to change the color scheme in Sketcher to work around the red-green axis cross to a non similar color of dimensions and fully constrained sketch. This results in an odd behavior, the constrained sketch reverts to green but if I continue to draw elements the green will change to my chosen white.
3.  If the axis cross needs be in sketcher for some reason, would it be possible to change the line width to .5, I have changed the color a very light green and red which does help with contrast. Not seeing a option to change the line width of the axis cross in preferences. 
I know you need to pick your battles with your development work and the default sketcher color scheme may not be one of them.
Thanks.

jomanxok, 2025-04-24:
I did get sketcher Appearance settings to work as it should, along with color change on the axis cross to a lighter shade. I will work with this, you have other things to think about.

PaddleStroke, 2025-04-25:
There is a lot of information in your feedbacks !
- For the layout, I understand. If the task panel was on the left it would avoid mouse going back and forth. Currently I would suggest to you to grab the tree and drop it on the left side, but not as an overlay widget. Then put the task panel in the left overlay. Like so : 

[image]

I think this layout is better for large screens. I was thinking to add in the setup wizard an option to select either this layout (for big screens) or the current one (for smaller screens)

For the sketch axis I seem to recall there is an open issue on freecad tracker. And perhaps even someone already working on it.

jomanxok, 2025-04-25:
Thanks, I do like the panel next to the tree and use it in that location a lot. Just to have it remember location and size is nice when a floating panel.

As for the sketcher cross really should have a toggle off. I did work out a color scheme and lowered the color value of the cross so it does not interfere with lines and dimensions. Also here is a link you may find interesting.
https://en.wikipedia.org/wiki/Solarized

From: PaddleStroke
Sent: Friday, April 25, 2025 9:05 AM
To: AstoCAD/FreeCAD
Cc: jomanxok; Author
Subject: Re: [AstoCAD/FreeCAD] AstoCAD Sketcher toolbar (Issue #22)

PaddleStroke left a comment (AstoCAD/FreeCAD#22)
There is a lot of information in your feedbacks !
• For the layout, I understand. If the task panel was on the left it would avoid mouse going back and forth. Currently I would suggest to you to grab the tree and drop it on the left side, but not as an overlay widget. Then put the task panel in the left overlay. Like so :
image.png (view on web)
I think this layout is better for large screens. I was thinking to add in the setup wizard an option to select either this layout (for big screens) or the current one (for smaller screens)
For the sketch axis I seem to recall there is an open issue on freecad tracker. And perhaps even someone already working on it.

jomanxok, 2025-04-26:
I made a short video of the panel docking in Fusion, I think this would work very well for FC. Replace the comment panel with the property view panel but keep the expand UP to maintain maximum length for tree view. I think restricting docking options would be very helpful for new users.  I realize that AutoDesk has million invested in development cost so they should come up with some good ideas, as least they did a good job with panel docking! I realize this coding is complicated so do what you can you have advanced FC a lot already.
Planning to let go of my subscription in November, the way FC is progressing that be should be just about right. I appreciate all the work you do, but remember to take care of yourself and do what’s right for you.

Regards 
Joe

P.S. 
Just a retired hobbist.
Hope the video quality is good enough, it’s the free version.

From: PaddleStroke
Sent: Friday, April 25, 2025 9:05 AM
To: AstoCAD/FreeCAD
Cc: jomanxok; Author
Subject: Re: [AstoCAD/FreeCAD] AstoCAD Sketcher toolbar (Issue #22)

PaddleStroke left a comment (AstoCAD/FreeCAD#22)
There is a lot of information in your feedbacks !
• For the layout, I understand. If the task panel was on the left it would avoid mouse going back and forth. Currently I would suggest to you to grab the tree and drop it on the left side, but not as an overlay widget. Then put the task panel in the left overlay. Like so :
image.png (view on web)
I think this layout is better for large screens. I was thinking to add in the setup wizard an option to select either this layout (for big screens) or the current one (for smaller screens)
For the sketch axis I seem to recall there is an open issue on freecad tracker. And perhaps even someone already working on it.