AstoCAD/FreeCAD #23: AstoCAD Windows build doesn't have installer

PaddleStroke, 2025-02-28:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

AstoCAD Windows build doesn't have installer

### Full version info

```shell
1.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

CIKoolK, 2025-03-01:
As you have found there is no formal classic _Windows_ "installer" (".exe" or ".msi"); but all you have to do is:

1. Copy the downloaded archive package to _C:/Program Files_;
2. Extract the single folder it contains (it will be named "AstoCAD" something or other;
3. Create a desktop shortcut with _C:/Program Files/[AstoCad...]/bin/AstoCAD.exe_ as the executable;
4. Name it _AstoCAD_ or whatever floats your boat;

PaddleStroke, 2025-12-04:
Fixed, installer now available.

closed 2025-12-04 by PaddleStroke