AstoCAD/FreeCAD #24: Sketcher Trim line doing a strange thing

paulgcrabb, 2025-03-02:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Trim line doing a strange thing when line is intersecting 2 concentric circles and is a tangent to the inner circle, the line is not fully constrained, the circles are. The attached video shows the behaviour.

https://github.com/user-attachments/assets/1b81fb87-6bb1-4bbf-8bfa-8c218b96f561

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40208 (Git) Conda
Build type: Release
Branch: AstoCAD
Hash: 760cbe471b1cf8d7408e7cbf48f876960012a59c
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/Light/Fusion
Logical/physical DPI: 96/93.2987
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

PaddleStroke, 2025-03-03:
This is a FreeCAD's known regression following Sketcher's refactoring efforts.
I'm not sure which issue it is because there are a lot of regression issues that have been opened recently.
But if you check https://github.com/FreeCAD/FreeCAD/issues?q=is%3Aissue%20state%3Aopen%20trim you will find several that looks similar to yours.

@AjinkyaDahale you may be interested to see this case of Trim failure perhaps?

I'll close the issue here since it's already opened in FC main and will likely be fixed soon.

closed 2025-03-03 by PaddleStroke

paulgcrabb, 2025-03-03:
Hmm... that's a lot of trim issues to resolve.

Apologies for posting this in the wrong place but I couldn't immediately see a similar case in freecads bug reports.

AjinkyaDahale, 2025-03-03:
@paulgcrabb it's the same kind of issue: a constraint changes into something it shouldn't. If you want, feel free to create the issue again in freecad (after confirming it happens there). I expect to still see this issue through the mention, and even if not, since I know of it's existence, I hope to eventually get to it.