AstoCAD/FreeCAD #25: Reset all preferences loose AstoCAD's preferences

PaddleStroke, 2025-03-03:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Reset all preferences loose AstoCAD's preferences

### Full version info

```shell
1.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

PaddleStroke, 2025-03-06:
The issue is that it is clearing preferences to FreeCAD's default. Not to AstoCAD defaults.
So if you clear preferences and open back astocad then it looks all messed up.

CIKoolK, 2025-03-06:
I "see", I misunderstood... Try making the .cfg files read-only unless you wish to make changes

On Linux they are located at: _$HOME/.config/AstoCAD_
On Windows at: _C:\Users\[username]\Application Data\AstoCAD_

PaddleStroke, 2025-03-06:
I found a solution to restore AstoCAD's default. It will be in next build

CIKoolK, 2025-03-06:
Cool...

PaddleStroke, 2025-03-19:
The problem is that the automatic build workflow on github is broken upstream in freecad since they updated to c++20. Someone is working on it but meanwhile we are stuck on 40545.
Meawhile to reset to AstoCAD preferences, what you can do is close AstoCAD , delete user.cfg and open AstoCAD again.

jomanxok, 2025-03-21:
I have it working with .cfg delete.