AstoCAD/FreeCAD #26: Unit is cut off while dimensioning

Roycinger, 2025-03-05:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When creating a shape with the related dimension, the unit (in my case "mm") is cut off in the view.

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40544 (Git) Conda
Build type: Release
Branch: AstoCAD
Hash: 3fef319941dbcfe7ebeebe2d2488c5c58704f212
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/Dark/Fusion
Logical/physical DPI: 96/81.6656
```

### Subproject(s) affected?

None

### Anything else?

[image]

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

PaddleStroke, 2025-03-05:
Found it and opened PR to core https://github.com/FreeCAD/FreeCAD/pull/20008/files
Will be in next builds of AstoCAD

PaddleStroke, 2025-04-18:
Build 41331 posted include this fix.

closed 2025-04-18 by PaddleStroke