AstoCAD/FreeCAD #28: Feature Request - Property View autohide

toskium, 2025-03-05:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In AstoCAD, the property view is positioned to the left of the model tree, which I find to be an excellent location. However, once enabled, it occupies a large portion of the screen. Disabling and re-enabling it can be somewhat annoying. Would it be possible to implement an autohide feature, so that the property view only appears when the mouse cursor moves to the left edge of the screen, making it faster to access?

[image]

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40208 (Git) Conda
Build type: Release
Branch: AstoCAD
Hash: 760cbe471b1cf8d7408e7cbf48f876960012a59c
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/Germany (en_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/Dark/Fusion
Logical/physical DPI: 96/92.6073
Installed mods: 
  * fasteners 0.5.37
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

PaddleStroke, 2025-03-05:
That is an interesting idea. I'll think on it
Meanwhile you have the shortcut 'alt + return' to open the view.
You can also right click an object and select properties to show the view

CIKoolK, 2025-03-05:
On my Linux system hovering the mouse pointer over the right border of the Property View panel will allow "dragging" to resize the width of the panel

[image]

toskium, 2025-03-06:
> On my Linux system hovering the mouse pointer over the right border of the Property View panel will allow "dragging" to resize the width of the panel

@CIKoolK yes, that works on the Windows version as well, unfortunately the Property View panel can't be smaller than what you can see in your screenshot, so it's not "out of the way" and in addition (at least for me) it needs to have a certain width to properly use Variable Sets (descriptive variable names take up more space).

mno13, 2025-03-11:
Would be possible to also add "Auto Close" option for Property View? When working on smaller screen (laptop), I prefer to open property view as separate window (alt+enter) , but it needs manual closing or it sits empty in the middle of screen when user deselects current item, which is inconvenient.

Adding big "CLOSE" button, when property view is undocked, would be also nice feature. "X" icon size is too small.

PaddleStroke, 2025-11-29:
still relevant

vanessa-luna, 2026-11-11:
What if we simply added a Std_ToggleLeftPanels like we have for Std_ToggleBottomPanels?

This way we can give a shortcut to it and then let us decide how to customize it?

[image]
[image]