AstoCAD/FreeCAD #30: Cannot install Sheetmetal

hptsys, 2025-03-09:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Same as, [](https://github.com/FreeCAD/FreeCAD/issues/19380#issuecomment-2649116031), but since AstoCad is wrapped into AppImage, I cannot manually install the missing netwrokx package. 

### Full version info

```shell
OS: Ubuntu 22.04.5 LTS (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.1.0dev.40545 (Git) Conda AppImage
Build type: Release
Branch: AstoCAD
Hash: 5b4bec68404d884ac9dcab7183f357d7e1585c07
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/Australia (en_AU)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/Dark/Fusion
Logical/physical DPI: 96/92.4513
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

PaddleStroke, 2025-03-11:
> Same as, , but since AstoCad is wrapped into AppImage

Sorry I cannot understand. A few things : 

- Sheetmetal should actually be pre-installed in AstoCAD (unless there's a problem).
- I don't know how linux addon management is done but you should be able to install addons even though it's an AppImage. FreeCAD's an AppImage as well. Is there something different from a FreeCAD weekly build that generates your problem?

hptsys, 2025-03-12:
Hi,
Yes, I can confirm Sheetmetal is already installed in AstoCAD.
The problem began when I tried to re-install it (just clicked install button on the add-on page). It warned networkx python dependency is needed. However, trying to resolve the dependency will fail.