AstoCAD/FreeCAD #31: Inability to create New Part

mno13, 2025-03-11:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

"create new part" is gone from global toolbar, no way to create Part container from Part Design / Part / Sheet Metal WB

It is only possible via Assembly Workbench during inserting new component.

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.40516 (Git) Conda
Build type: Release
Branch: AstoCAD
Hash: c44e29ee0bca1be812e1282106596a245b7c42f4
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/Light/Fusion
Logical/physical DPI: 96/161.377
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

PaddleStroke, 2025-03-13:
There's 'New Part' under the 'New File' command:

[image]

But this new command does not let you create a part container in the current file. Which was intentional since it's trying to enforce the 1 part per document rule.

But this is quite limiting, so I have just finished reworking this tool so that now it looks like this : 

[image]

It will be in next builds of AstoCAD. But the github workflow to create the build is broken since upstream moved to C++20. So it will take some time before that works again.

mno13, 2025-03-13:
Easy to oversight, as for some people default action for new document is "ctrl+n".

Could it be made workbench sensitive? In Part/PartDesign WB there would be "New Part" as main action, in Asm "New Assembly", in Tech Draw "New Drawing" etc.?

PaddleStroke, 2025-04-18:
Hmm I'm not sure. I think ultimately it would be best to move people into using '1 part per file' or '1 assembly per file' similarly to SW is doing. Which is why I moved the New Part there.

closed 2025-04-18 by PaddleStroke