AstoCAD/FreeCAD #33: Jumping treeview on clocking

Footz355, 2025-03-13:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

While clicking on the treeview elements in the overlay, the whole treeview drops a bit to the bottom and back again. It does not happen outside of overlay.

[image]

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40545 (Git) Conda
Build type: Release
Branch: AstoCAD
Hash: 5b4bec68404d884ac9dcab7183f357d7e1585c07
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/Dark/Fusion
Logical/physical DPI: 96/141.585
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

PaddleStroke, 2025-03-13:
I see this is because you have the taskview in the same overlay as the treeview. Is it intentional?

It is caused by the automatic resize of the task panel, which is not perfect yet since it makes this jumping. Though in the default layout I have done it is not a problem since the taskview in by itself on the right.

closed 2025-03-13 by Footz355

PaddleStroke, 2025-03-13:
I am fixing this issue anyway since that should not be happening.

Another idea of layout that is good especially if you have a big screen is this one : 

[image]

I really like the taskbox being closer to the rest. The only downside is for small screens.