AstoCAD/FreeCAD #36: Can't select much of anything since the beginning.

Leemstradamus, 2025-03-25:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Hello, I signed up when this project was announced and have had nothing but issues. The build that I downloaded it wont allow me to select anything if the window is maximized. In the screen shot you can see the dropdown but on my screen at the time it doesn't show at all. 

https://github.com/user-attachments/assets/259b7ca0-35ef-48d4-880d-8d46559a0dfe

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40208 (Git) Conda
Build type: Release
Branch: AstoCAD
Hash: 760cbe471b1cf8d7408e7cbf48f876960012a59c
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/Dark/Fusion
Logical/physical DPI: 96/108.933
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

PaddleStroke, 2025-03-28:
Hello @Leemstradamus 
Sorry about this issue.
So this bug happens only when the window is maximized right?
Can you please try to download the latest build 40545 from the website and see if you still have this bug?

Thanks

Leemstradamus, 2025-03-28:
I've sent you two emails about not being able to download anything. I subscribed again and have the PayPal transaction but it isn't reflected on my astocad account.

PaddleStroke, 2025-03-28:
Can you please send me an email with the transaction so that I'll check it out ? (unless if it is you Roland who just sent me an email)

Leemstradamus, 2025-03-28:
I sent one yesterday with a screenshot of the paypal transaction to hello@8bcraft.com at 09:39:26 -0700. I can send it again if this isn' the correct address.

PaddleStroke, 2025-03-28:
Found it. Sorry I have not checked this email since Monday. It is unfortunate that paypal is showing this email rather than hello@astocad.com. I need to find a solution for this.
I'll fix your situation asap.

Leemstradamus, 2025-03-28:
> Found it. Sorry I have not checked this email since Monday. It is unfortunate that paypal is showing this email rather than [hello@astocad.com](mailto:hello@astocad.com). I need to find a solution for this. I'll fix your situation asap.

Thank you I appreciate it.