AstoCAD/FreeCAD #37: SpaceMouse

jomanxok, 2025-03-28:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

SpaceMouse will only work with one instance of FC or AstoCAD, which ever was opened first.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40545 (Git) Conda
Build type: Release
Branch: AstoCAD
Hash: 5b4bec68404d884ac9dcab7183f357d7e1585c07
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/Dark/Fusion
Logical/physical DPI: 96/127.628
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/127.628
```

### Subproject(s) affected?

None

### Anything else?

I still have mouse control, but only one with SpaceMouse control.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

jomanxok, 2025-04-01:
> What happens if you open 2 instances of FreeCADs?

The 3dconnexion mouse will only work with the FC instance that is opened
first, the regular mouse works as expected in both.

Daniel-Steinberger, 2025-09-09:
I cannot reproduce this. For me the behavior (Windows 11) is as follows: Spacemouse works in any number of instances, as long as the executable name is "freecad.exe". It does not work, when the "astocad.exe" ist launched.

My guess is, that the driver only enables the functionality when it recognizes the Application by name.
That's why I launch Astocad by the freecad.exe Binary provided in the bin Directory.

PaddleStroke, 2025-09-09:
> as long as the executable name is "freecad.exe"

Thanks for this clue. I will investigate

PaddleStroke, 2025-11-29:
Still relevant