AstoCAD/FreeCAD #41: Include hash with download links

OfficialKris, 2025-04-08:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

An MD5 hash next to each download link would be useful for verifying file integrity after download. There is currently no way to do this. This could lead to corruption in the program.

### Full version info

```shell
OS: EndeavourOS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.40545 (Git) Conda AppImage
Build type: Release
Branch: AstoCAD
Hash: 5b4bec68404d884ac9dcab7183f357d7e1585c07
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/Light/Fusion
Logical/physical DPI: 144/201.061
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

PaddleStroke, 2025-04-09:
Thanks for the report. Yes I can add the hash files in the future. I will add them when I upload the next build.

PaddleStroke, 2025-04-18:
SHA256 in text files have been added now to the account page

closed 2025-04-18 by PaddleStroke