AstoCAD/FreeCAD #44: Windows: I can move to another monitor but lose the ability to click anything in the top right of the window.

chris99336, 2025-04-21:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When Astocad is moved to another monitor, everything in the top right is unclickable. This includes the workbenches and even the close button. I thought maybe it was the new build, but it occurs in the previous build as well.

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.41331 (Git)
Build date: 2025/04/18 06:30:41
Build type: Release
Branch: AstoCAD
Hash: 3da2f1a7272c5d93beccb3147db3c66a9a756f2c
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/93.2987/1.5
Installed mods: 
  * Curves 0.6.61
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

PaddleStroke, 2025-06-12:
Thanks for the report. I cannot replicate this on my laptop. I need to find another screen to test with my desk computer.
I wonder if it has something to do with the screen or the scale.
Are you using any Windows11 scaling for the primary or secondary monitors?

PaddleStroke, 2025-11-29:
To be tested in next release when the build workflow is fixed

chris99336, 2025-11-29:
> To be tested in next release when the build workflow is fixed

I forgot to get back to you. When I get back to work Monday I will try again on a multi-monitor setup as well and let you know. Most of the time I'm at home on a single monitor.