AstoCAD/FreeCAD #45: [Feature request] Add optional solid view for clipping plane

mathew500, 2025-04-21:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When using "clipping plane" (under "view" options), the cut plane is viewed as shells, though the object being solids. Could an option be added to the "clipping plane" panel to enable "solid" view. Either it should remember last used state, or a default should be added to a preference page somewhere.
To get an idea of how the solid view would look like, one can achieve it by making a cube in "Part" workbench and do a boolean cut.

### Full version info

```shell
OS: Zorin OS 17.3 (zorin:GNOME/zorin/wayland)
Architecture: x86_64
Version: 1.1.0dev.41331 (Git) AppImage
Build date: 2025/04/18 06:30:41
Build type: Release
Branch: AstoCAD
Hash: 3da2f1a7272c5d93beccb3147db3c66a9a756f2c
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/143.907/1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

PaddleStroke, 2025-04-25:
That is a good feature request. I wanted to look into this. But I also wanted to make an easier tool for section views. I need to investigate the possibilities here.

R0flcopt3r, 2025-07-13:
I really like how Fusion 360 does it, where the cut faces have a different texture than normal. Would be great to have an option to turn that on as well.