AstoCAD/FreeCAD #47: [Linux] Recent linux build doesn't want to run on x86_64

xphil, 2025-04-23:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Recent Linux build of AstoCAD doesn't want to execute on x86_64 Ubuntu 24.04.

`[phobos][freecad] $ ./AstoCAD_41331-Linux.AppImage 
-bash: ./AstoCAD_41331-Linux.AppImage: cannot execute binary file: Exec format error
[phobos][freecad] $ sha256sum AstoCAD_41331-Linux.AppImage 
09d26c37fcba989c102e32f7667a62e07a1183b66f91005863c42638016c00e1  AstoCAD_41331-Linux.AppImage
[phobos][freecad] $ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 24.04.2 LTS
Release:        24.04
Codename:       noble
`

### Full version info

```shell
09d26c37fcba989c102e32f7667a62e07a1183b66f91005863c42638016c00e1  AstoCAD_41331-Linux.AppImage
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

PaddleStroke, 2025-04-25:
Could you please confirm if the current FreeCAD weekly version is working correctly for you? I have not made any conscious choice regarding this so it may be a upstream issue. Or perhaps I did something wrong during rebase of the feedstock.

xphil, 2025-04-25:
Just tried the latest FreeCAD weekly build for Linux x86_64 - FreeCAD_weekly-2025.04.21-Linux-x86_64-py311.AppImage, also tried previously saved weekly FreeCAD_weekly-builds-41264-conda-Linux-x86_64-py311.AppImage - both work perfectly.

When I download Linux build from your website, the file doesn't have architecture in its name, only OS.
The "file" linux utility shows that it is, in fact, Linux binary, BUT compiled for AArch64 instead of x86_64, hence I can't run it on my Intel-based Linux machine.

pellcorp, 2025-04-26:
I have same issue

PaddleStroke, 2025-04-27:
Ok so I must have had a problem when rebasing the feedstock fork of FreeCAD. I will investigate asap.

PaddleStroke, 2025-04-27:
I have updated the builds there is now one intel and one arm build for linux can you please test and report if it's OK?

xphil, 2025-04-27:
@PaddleStroke Linux-intel build works. Thanks for fixing it!

closed 2025-04-28 by PaddleStroke