AstoCAD/FreeCAD #48: [Feature Request] Make the Polyline tool’s keyboard shortcut configurable

toskium, 2025-04-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The new Polyline tool is excellent, but its shortcut is hard-wired to M, a key on the right-hand side of the keyboard. I remap all of my most-used shortcuts to the left half of the keyboard so I can keep my right hand on the mouse, but that isn’t possible for Polyline at the moment.

My only current workaround is to rebind M at the keyboard firmware level — an inelegant hack that affects other applications. If the Polyline shortcut were exposed in FreeCAD’s standard Customize > Keyboard settings, users could assign it to a key that better suits their workflow and ergonomics.

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.41331 (Git)
Build date: 2025/04/18 06:30:41
Build type: Release
Branch: AstoCAD
Hash: 3da2f1a7272c5d93beccb3147db3c66a9a756f2c
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/Germany (en_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * fasteners 0.5.37
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

PaddleStroke, 2025-11-29:
still relevant

vanessa-luna, 2026-11-11:
While this is not customizable yet, it has been changed to 'Alt' now instead of 'M' which is an improvement, but PgUp and PgDown for the other keys are not generally accessible keys on laptops or small keyboards