AstoCAD/FreeCAD #49: lost password does not work

bitbasher, 2025-04-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

i am trying to download the latest version of the Astocad package for windows but I am not able to login on the astocad site. My password is not being accepted .. i tried to do the "lost password" thing for email address
 jeff at jartisan dot com 
and it said that it would send an email with a link .. nothing ever arrived.

I am a fully paid up annual member 

Can anyone send me a link to the download and email me a link to reset my password on astocad.com please?

I have not been able to find any method for directly contacting the developer so i am hoping a post here will be noticed.

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.40208 (Git) Conda
Build type: Release
Branch: AstoCAD
Hash: 760cbe471b1cf8d7408e7cbf48f876960012a59c
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/Belgium (en_BE)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/Light/Fusion
Logical/physical DPI: 96/91.7938
Installed mods: 
  * 3D_Printing_Tools
  * addFC 1.4.3
  * boltsfc 2022.11.5
  * btl 0.9.9
  * CurvedShapes 1.0.13
  * Curves 0.6.58
  * Defeaturing 1.2.3
  * DynamicData 2.73.0
  * fasteners 0.5.37
  * freecad.gears 1.3.0
  * FreeGrid 2.2.0
  * Gridfinity 0.10.3
  * LCInterlocking
  * MeshRemodel 1.10.37
  * Pyramids-and-Polyhedrons
  * sheetmetal 0.7.21
  * symbols_library
  * ThreadProfile 1.92.0
  * toSketch 1.0.1
  * woodworking 0.21.2.33771.0.23.1
  * workfeature
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

PaddleStroke, 2025-04-25:
Hi Jeff,
Sorry about this issue. For those kind of issue the best is to reach by email at hello astocad dot com.
I have already sent you the email to reset the password. Can you confirm if you received it? If not then please send me an email and we'll figure out how to proceed.

closed 2025-04-25 by PaddleStroke

PaddleStroke, 2025-04-27:
Hi Jeff,
I received your emails correctly. I confirm that your account is using jartisan dot com.

I replied to your messages, can you see this reply? If not then perhaps my emails are in spam?

PaddleStroke, 2025-04-29:
I just tried to send you the reset email to the telenet address. Did you received this one?

bitbasher, 2025-05-05:
i don't know if this is really fixed .. i did get the link in a New Password email and for me the login issues was solved .. but if you sent that email manually the Lost Password page on the AstoCAD site may still NOT be working. I note that i used that page more than 10 times and _never_ got an email .. until the one that (i think) you send manually

PaddleStroke, 2025-05-06:
I think I fixed it actually. The issue was probably that the admin email of the website was my personal gmail address and not a @astocad.com email. So some email provider would block the email as spam. I'm not sure but from my research it is the n1 reason why reset email are not received.