AstoCAD/FreeCAD #54: AstoCAD crash on trimming operations (macOS)

mariogt, 2025-05-03:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

AstoCAD crashed randomly when I tried to trim some geometry on the Sketch, this error is shown when I hover over the section that I want to trim:

Unhandled unknown C++ exception in ViewProvider::eventCallback (Event type: SoLocation2Event, object type: SketcherGui::ViewProviderSketch)

Then if I click to trim that section the App crashes.

Note: I just tried the latest weekly build of FreeCAD 41488 and works fine on trim operations.

Attached is the error dump

[dump.txt](https://github.com/user-attachments/files/20021824/dump.txt)

### Full version info

```shell
OS: macOS Sequoia (15.4)
Architecture: arm64
Version: 1.1.0dev.41331 (Git)
Build date: 2025/04/24 09:25:36
Build type: Release
Branch: AstoCAD
Hash: 88f5b90eb2bc8f1a2cdc3b572df3a604c4d9aa70
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/Chile (en_CL)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/Dark/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
Installed mods: 
  * Render
  * fasteners 0.5.38
  * 3D_Printing_Tools
  * Curves 0.6.61
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

PaddleStroke, 2025-05-05:
This is most likely a upstream issue. Since some sketcher refactor work, trim has been full of bug. If that does not happen in weekly build 41488 then perhaps a fix has been merged between current astocad build and currently weekly build.

Let's see if this bug is still here in next astocad build.

mtancoigne, 2025-05-16:
I have the same behavior on Debian 12 with the appImage (git hash 3da2f1a7272c5d93beccb3147db3c66a9a756f2c);

The easiest reproduction steps are as follow:
1. Create new part
2. Create a sketch
3. Draw a circle intersecting a rectangle
4. Trim 
5. Crash.

[image]

It works like a charm on FreeCAD 1.0.0 (git hash 2fcc5317fe3aee96ca73475986a577719fc78e20, appImage version)

PaddleStroke, 2025-05-16:
Yes the regression was introduced after 1.0 so it makes sense that it worked then. And the PR fixing this was merged already.
The problem is that I currently can't get the windows build to work. The github workflow appears to be broken upstream again.

I can still push the new linux and mac builds. I'll do that tomorrow

mtancoigne, 2025-05-22:
I tested with the latest appImage (build 41726) and the issue seems to be gone (at least, I can reproduce the few steps without a crash :))

PaddleStroke, 2025-05-22:
Great so I will close this then :)

closed 2025-05-22 by PaddleStroke