AstoCAD/FreeCAD #55: Release Notes Location?

vanessa-luna, 2025-05-05:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Hi, aside from the main page on AstoCAD.com I cannot find a release notes file.

I just upgraded from 40545 to 41331 and would like to know what changes were made. Is there any way to access this information?

I apologize if this information is actually easy to find. 

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41331 (Git)
Build date: 2025/04/24 09:25:36
Build type: Release
Branch: AstoCAD
Hash: 88f5b90eb2bc8f1a2cdc3b572df3a604c4d9aa70
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/134.063/1.5
Installed mods: 
  * A2plus 0.4.68
  * Manipulator 1.5.8
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

PaddleStroke, 2025-05-06:
I have not yet made a detailed release note actually. The situation is being a bit complex because I am pushing some things (bug fixes mostly so far) into main. And I also rebase on main so there's a lot of new stuff coming from main too.
From now on I will keep a release notes file and publish it with new builds.

vanessa-luna, 2025-05-10:
I understand there is a lot to manage and your focus is on the work. The right balance between time on code and time on documentation is hard to strike. Regardless, Asto has finally helped me launch into FreeCAD and so far I'm pretty happy. There are a lot of nice features compared to F360.