AstoCAD/FreeCAD #56: [Fedora, KDE] Window is always maximized

R0flcopt3r, 2025-05-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

If I drag the "titlebar" (top of the AstoCAD window) nothing happens. If i click the button to maximize/restore nothing happens. The window always cover my entire screen. I can minimize and close just fine.

If i disable `Custom title bar` everything is working correctly again, but it looks very ugly.

### Full version info

```shell
OS: Fedora Linux 42 (KDE Plasma Desktop Edition) (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.1.0dev.41331 (Git) AppImage
Build date: 2025/04/24 09:25:36
Build type: Release
Branch: AstoCAD
Hash: 88f5b90eb2bc8f1a2cdc3b572df3a604c4d9aa70
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Norwegian Bokmål/Norway (nb_NO)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/109.338/1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

CIKoolK, 2025-05-18:
The "Custom title bar" only works on Windows.

[That is presented here](https://www.astocad.com/index.php/2025/02/12/custom-title-bar/).

I personally don't like it--don't know if that's 'cause I'm old (77) or my fundamental distaste for anything having to do with M$...

R0flcopt3r, 2025-05-18:
Then it should be disabled on Linux and macOS until it has been fixed. Made for a pretty bad first impression.

CIKoolK, 2025-05-18:
I agree, however it could go away permanently (or even fall into "The Pit") and I'd not care--with my geezer eyesight (diabetic retinopathy from lifelong diabetes)--I like plain ol' title bars.

I find it odd at times that 35 years after passage of ADA, and with all the leftist wringing of hands and bed-wetting over the "rights" of various groups; so much of this trendy UI stuff is unusable to me. "Videos", both informative and tutorial, for example, please just give me a printed/illustrated report!

PaddleStroke, 2025-06-02:
It seems this feature is a bit tricky to get to work on all linux plateforms. On macOS and some linux it is working ok, although not as great as on windows. So it is probably something in your setup that is different. I will need to investigate.
Luckily Qt 6.9 is supposed to introduce native support for frameless windows. So perhaps all of these trouble will be gone.

CIKoolK, 2025-06-02:
No need to prioritize this on my account, due to my eyesight issues (diabetic retinopathy) I like visible frames...

PaddleStroke, 2025-06-02:
I don't mean your setup has something special. I just meant it may be specific to Fedora.
And yes I agree it needs to be properly supported.

fronald, 2025-07-14:
I'm using Fedora 42 with KDE 6 and works. You can try export some environment variables like:

```sh
#!/usr/bin/bash
export QT_QPA_PLATFORM=xcb
export LD_PRELOAD=/usr/lib64/libdrm_amdgpu.so.1
/usr/local/apps/AstoCAD_42575-Linux-intel.AppImage $@
```

https://github.com/user-attachments/assets/a93fc72b-fe1e-43de-8452-998297feb274

CIKoolK, 2025-07-14:
> I don't mean your setup has something special. I just meant it may be specific to Fedora. And yes I agree it needs to be properly supported.

Please do not--in your zeal to be trendy--do away with the classic title-bar--due to my eyesight I would have to abandon AstoCAD if that were to happen...