AstoCAD/FreeCAD #59: crashes after confirming mirror in transform tool

knezi, 2025-06-01:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Create a part design object and add multitransform, first a linear array of a single feature, then try to mirror the linear pattern. After clicking ok in the mirror tool, it crashes.

Can be reproduced in the attached file, just open the multitransform tool and add mirror (I leave the default vertical plane) and click ok.

```
(qt.qpa.plugin) Could not find the Qt platform plugin "wayland" in ""
/home/runner/work/AstoCAD-Bundle/AstoCAD-Bundle/conda/linux/AppDir/usr/share/libdrm/amdgpu.ids: Adresář nebo soubor neexistuje
Program received signal SIGSEGV, Segmentation fault.
#0  /lib64/libc.so.6(+0x1a050) [0x7faf23a3a050]
#1  0x7faf25486bd1 in QComboBox::clear() from /tmp/.mount_astocaaCNCnH/usr/bin/../lib/libQt6Widgets.so.6+0x51
#2  0x7faf2765c22c in Gui::ComboLinks::clear() from /tmp/.mount_astocaaCNCnH/usr/bin/../lib/libFreeCADGui.so+0x2c
#3  0x7faf2765c2ca in Gui::ComboLinks::~ComboLinks() from /tmp/.mount_astocaaCNCnH/usr/bin/../lib/libFreeCADGui.so+0xa
#4  0x7fae883be36e in PartDesignGui::TaskMirroredParameters::~TaskMirroredParameters() from /tmp/.mount_astocaaCNCnH/usr/lib/PartDesignGui.so+0xae
#5  0x7fae883be3da in PartDesignGui::TaskMirroredParameters::~TaskMirroredParameters() from /tmp/.mount_astocaaCNCnH/usr/lib/PartDesignGui.so+0xa
#6  0x7fae883c58f2 in PartDesignGui::TaskMultiTransformParameters::closeSubTask() from /tmp/.mount_astocaaCNCnH/usr/lib/PartDesignGui.so+0x202
#7  /tmp/.mount_astocaaCNCnH/usr/bin/../lib/libQt6Core.so.6(+0x31684a) [0x7faf2411684a]
#8  /tmp/.mount_astocaaCNCnH/usr/bin/../lib/libQt6Widgets.so.6(+0x47bf53) [0x7faf2547bf53]
#9  0x7faf2547c2df in QAbstractButton::mousePressEvent(QMouseEvent*) from /tmp/.mount_astocaaCNCnH/usr/bin/../lib/libQt6Widgets.so.6+0xdf
#10  0x7faf255c388c in QWidget::event(QEvent*) from /tmp/.mount_astocaaCNCnH/usr/bin/../lib/libQt6Widgets.so.6+0xa8c
#11  0x7faf255b3e48 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_astocaaCNCnH/usr/bin/../lib/libQt6Widgets.so.6+0x88
#12  0x7faf255b4cbd in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_astocaaCNCnH/usr/bin/../lib/libQt6Widgets.so.6+0xdad
#13  0x7faf27164139 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_astocaaCNCnH/usr/bin/../lib/libFreeCADGui.so+0xc9
#14  0x7faf2415fb8a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_astocaaCNCnH/usr/bin/../lib/libQt6Core.so.6+0x17a
#15  0x7faf255bbe7c in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /tmp/.mount_astocaaCNCnH/usr/bin/../lib/libQt6Widgets.so.6+0x1ec
#16  /tmp/.mount_astocaaCNCnH/usr/bin/../lib/libQt6Widgets.so.6(+0x5be8c7) [0x7faf255be8c7]
#17  /tmp/.mount_astocaaCNCnH/usr/bin/../lib/libQt6Widgets.so.6(+0x5c5440) [0x7faf255c5440]
#18  0x7faf255b3e48 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_astocaaCNCnH/usr/bin/../lib/libQt6Widgets.so.6+0x88
#19  0x7faf27164139 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_astocaaCNCnH/usr/bin/../lib/libFreeCADGui.so+0xc9
#20  0x7faf2415fb8a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_astocaaCNCnH/usr/bin/../lib/libQt6Core.so.6+0x17a
#21  0x7faf247d1970 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /tmp/.mount_astocaaCNCnH/usr/bin/../lib/libQt6Gui.so.6+0x810
#22  0x7faf24831e14 in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_astocaaCNCnH/usr/bin/../lib/libQt6Gui.so.6+0xb4
#23  /tmp/.mount_astocaaCNCnH/usr/lib/qt6/plugins/platforms/../../../libQt6XcbQpa.so.6(+0x58f90) [0x7faf11405f90]
#24  /tmp/.mount_astocaaCNCnH/usr/bin/../lib/./libglib-2.0.so.0(+0x5b5e6) [0x7faf227bf5e6]
#25  /tmp/.mount_astocaaCNCnH/usr/bin/../lib/./libglib-2.0.so.0(+0x5e7b7) [0x7faf227c27b7]
#26  /tmp/.mount_astocaaCNCnH/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x7faf227c2ece]
#27  0x7faf23ef6b12 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_astocaaCNCnH/usr/bin/../lib/libQt6Core.so.6+0x72
#28  0x7faf2419747c in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_astocaaCNCnH/usr/bin/../lib/libQt6Core.so.6+0x25c
#29  0x7faf241977da in QCoreApplication::exec() from /tmp/.mount_astocaaCNCnH/usr/bin/../lib/libQt6Core.so.6+0x9a
#30  0x7faf270be0b6 in Gui::Application::runApplication() from /tmp/.mount_astocaaCNCnH/usr/bin/../lib/libFreeCADGui.so+0x756
#31  /tmp/.mount_astocaaCNCnH/usr/bin/AstoCAD(+0x6d34) [0x5613f4719d34]
#32  /lib64/libc.so.6(+0x3248) [0x7faf23a23248]
#33  /lib64/libc.so.6(__libc_start_main+0x8b) [0x7faf23a2330b]
#34  /tmp/.mount_astocaaCNCnH/usr/bin/AstoCAD(+0x703b) [0x5613f471a03b]
```

[test.tar.gz](https://github.com/user-attachments/files/20539342/test.tar.gz)

### Full version info

```shell
OS: Fedora Linux 41 (Sway) (sway/sway/wayland)
Architecture: x86_64
Version: 1.1.0dev.41851 (Git) AppImage
Build date: 2025/05/23 13:05:39
Build type: Release
Branch: AstoCAD
Hash: f3061debedacda10df4b447d9dedc891c38a6c64
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Czech/Czechia (cs_CZ)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/127.23/1
Installed mods: 
  * fasteners 0.5.38
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

knezi, 2025-06-01:
I've just tried it in FreeCAD 1.0.1 and it doesn't crash.

PaddleStroke, 2025-06-17:
Should be fixed in released build 42131.

closed 2025-06-17 by PaddleStroke