AstoCAD/FreeCAD #60: Pocket Operation Fails in AstoCAD but Works in FreeCAD 1.0

jsittner, 2025-06-10:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The pocket operation no longer works in AstoCAD, even though the same operation functions correctly in FreeCAD 1.0.

[image]

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.41851 (Git)
Build date: 2025/05/23 13:05:39
Build type: Release
Branch: AstoCAD
Hash: f3061debedacda10df4b447d9dedc891c38a6c64
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: TinkerCAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/73.4044/1.25
Installed mods: 
  * SearchBar 1.4.2
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

PaddleStroke, 2025-06-12:
Thanks for the report. Is it specific to UpToFace or is it all modes that fail?
In 41851 I don't think I actually made any changes to pocket yet. Perhaps main was broken at this point. I will check this out and release a new build soon anyway.

jsittner, 2025-06-17:
Build 42131 resolved the issue. Thank you! I also noticed the additional functionality in the pad tool and really appreciate the enhancements.