AstoCAD/FreeCAD #79: New Symbol Tool: SVG Points not Coincident

Daniel-Steinberger, 2025-08-05:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Hey, when using your new symbol tool, often a resulting Sketch cannot be padded/pocketed, because points are not coincident.

I've been helping myself with selecting the Sketch and running `obj.makeMissingPointOnPointCoincident(True)` there.
Usually it will fix most of the problems.
It would however be great, if this could be done automatically (or on demand via the gui).

The same issue arises with some fonts. This could help in that situation, too.

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.42582 (Git)
Build date: 2025/07/16 09:33:02
Build type: Release
Branch: AstoCAD
Hash: 7725e5b2b314006c48ec2551ef32c0738079597d
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: German/Germany (de_DE)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/107.584/1.5
Installed mods: 
  * FusedFilamentDesign 0.25.250
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

PaddleStroke, 2025-08-06:
When I build the library of basic logos for this feature, I noticed that almost all of the SVG have small flaws that made them not-closed. So I had to edit them all manually with inkscape to fix the SVG so that they would work out of the box.

But your idea is great, I can indeed run something like `obj.makeMissingPointOnPointCoincident(True)` to fix bad svgs.

Daniel-Steinberger, 2025-09-02:
```
obj.detectMissingPointOnPointConstraints()
obj.analyseMissingPointOnPointCoincident()
obj.makeMissingPointOnPointCoincident(True)
```

PaddleStroke, 2026-11-16:
Will be fixed in next build (in a few days)