AstoCAD/FreeCAD #80: "The selected asset path is not writable" when editing settings

markus-seidl, 2025-08-06:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When I'm editing any preferences (even without chaning) and "Apply" or "OK" the dialog, the message "The selected asset path is not writable" appears. Sometimes it even crashes the app completely.

The log shows this (can't copy and paste the log in bulk?):

[image]

### Full version info

```shell
OS: macOS 15.5
Architecture: arm64
Version: 1.1.0dev.42836 (Git)
Build date: 2025/07/29 06:28:12
Build type: Release
Branch: AstoCAD
Hash: 214716b2833df45ffb7798a82ae557f870886fc6
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/Germany (en_DE)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/Light/
Logical DPI/Physical DPI/Pixel Ratio: 72/111/1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

PaddleStroke, 2025-08-11:
Thanks for the report. I am not sure what can cause this regression. I need to investigate. Are other Mac users also facing this problem?

Also have you tried to re-download the application and check the SHA number to make sure it's good? Besides I see you are using the mac arm version, did you do the command as explained there : https://github.com/AstoCAD/FreeCAD/issues/21 to be able to open the application?

markus-seidl, 2025-08-11:
> Are other Mac users also facing this problem?
Dunno, it's my first time using Astocad at all. 
I think I've used FreeCad some time ago - maybe some property files from old FreeCad are causing problems?

> Besides I see you are using the mac arm version, did you do the command as explained there : https://github.com/AstoCAD/FreeCAD/issues/21 to be able to open the application?
Yes, I've done that.

> Also have you tried to re-download the application and check the SHA number to make sure it's good?
```
> Downloads sha256sum AstoCAD_42836-macOS-arm.dmg
e27b3771a600ddea0e0ffb94f7d4f280eec27ac68a4f8e9b04aefbaeba9d1e26  AstoCAD_42836-macOS-arm.dmg
```

PaddleStroke, 2025-08-17:
In your screenshots (in the other issue) I see that the AstoCAD preference pack is not applying in your computer. Is the fix that you found also fixing that? ie is AstoCAD looking like stock freecad or like the screenshot on astocad website?

MisterMakerNL, 2025-08-17:
I had the same issue here: https://github.com/FreeCAD/FreeCAD/pull/23046
In my opinion this is a bug, since if you open CAM wb it creates this directory. And you shouldn't need it if you are not using CAM.

PaddleStroke, 2025-08-20:
@markus-seidl 
As a workaround, here is a link to the astocad preference pack : https://github.com/AstoCAD/astocad-feedstock/blob/master/recipe/branding/AstoCAD-defaults.cfg

You can add it to the preference packs and apply it.

PaddleStroke, 2025-09-10:
Can you please check in latest version if this bug is still present?