AstoCAD/FreeCAD #87: AstoCAD Preference Pack isn't applied

jtyocum, 2025-08-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

After installing AstoCAD, it looks just like FreeCAD. It appears that none of the settings in https://github.com/AstoCAD/astocad-feedstock/blob/master/recipe/branding/AstoCAD-defaults.cfg are being applied. For example, the Mouse Navigation is still the FreeCAD default "CAD", all of the workbenches are enabled, etc. Perhaps related, in the notifications area are several messages related to Preferences (screenshot attached).

[image]

There is a similar report of the preference pack issue on Reddit, https://www.reddit.com/r/FreeCAD/comments/1mrj3cg/paid_for_astocad_but_looks_like_freecad_anybody/

### Full version info

```shell
OS: macOS 15.6
Architecture: arm64
Version: 1.1.0dev.42836 (Git)
Build date: 2025/07/29 06:28:12
Build type: Release
Branch: AstoCAD
Hash: 214716b2833df45ffb7798a82ae557f870886fc6
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/Dark/
Logical DPI/Physical DPI/Pixel Ratio: 72/108/2
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

indiego-gh, 2025-09-03:
Similar problem here, but it's only the 'size of the toolbar icons', that is not saved.
When opening the preferences dialog I directly get the following messages:

```
12:21:46  Cannot find icon: preferences-cam
12:21:46  Traceback (most recent call last):
12:21:46    File "V:\AstoCAD\Mod\CAM\Path\Main\Gui\PreferencesJob.py", line 40, in __init__
12:21:46      self.form = FreeCADGui.PySideUic.loadUi(":preferences/PathJob.ui")
12:21:46                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
12:21:46    File "<string>", line 4, in <module>
12:21:46  RuntimeError: Cannot open file
12:21:46  JobPreferencesPage is not a preference page
12:21:46  Traceback (most recent call last):
12:21:46    File "V:\AstoCAD\Mod\CAM\Path\Base\Gui\PreferencesAdvanced.py", line 35, in __init__
12:21:46      self.form = FreeCADGui.PySideUic.loadUi(":preferences/Advanced.ui")
12:21:46                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
12:21:46    File "<string>", line 4, in <module>
12:21:46  RuntimeError: Cannot open file
12:21:46  AdvancedPreferencesPage is not a preference page
12:24:26  Cannot find icon: preferences-cam
12:24:26  Traceback (most recent call last):
12:24:26    File "V:\AstoCAD\Mod\CAM\Path\Main\Gui\PreferencesJob.py", line 40, in __init__
12:24:26      self.form = FreeCADGui.PySideUic.loadUi(":preferences/PathJob.ui")
12:24:26                 
```

PaddleStroke, 2025-09-09:
@indiego-gh are you also on macOS?
I cannot replicate this issue on windows or linux and don't have a mac to test.

@jtyocum Did you used previous version of AstoCAD and / or freecad? If yes have you tried to clear the preferences before launching AstoCAD? AstoCAD and FreeCAD preferences should be separated, but maybe on mac they are not. Which could be the root cause.

jtyocum, 2025-09-17:
> [@jtyocum](https://github.com/jtyocum) Did you used previous version of AstoCAD and / or freecad? If yes have you tried to clear the preferences before launching AstoCAD? AstoCAD and FreeCAD preferences should be separated, but maybe on mac they are not. Which could be the root cause.

I had previously used FreeCAD. However, before switching to AstoCAD, I deleted the FreeCAD preferences files and folder.

zihotki, 2025-09-19:
> [@jtyocum](https://github.com/jtyocum) Did you used previous version of AstoCAD and / or freecad? If yes have you tried to clear the preferences before launching AstoCAD? AstoCAD and FreeCAD preferences should be separated, but maybe on mac they are not. Which could be the root cause.

I also encountered the same issue on my mac. I also had FreeCad installed. This is a new machine and I installed 1.1 development version of FreeCAD. Settings is something I checked first but it all looks correct and settings are separate between the apps, `/Users/USER/Library/Preferences/`:

```
AstoCAD/
AstoCAD.plist
FreeCAD/
freecad.plist
com.astocad.AstoCAD.plist
com.freecad.FreeCAD.plist
```

`/Users/USER/Library/Application Support/`:

```
FreeCAD/
AstroCAD/
```

I can only speculate that may be something went sideways during initial initialization. I can try to test/debug it if there are instructions available.

closed 2025-12-14 by github-actions[bot]