AstoCAD/FreeCAD #93: Group constraint is not copied with the Sketcher_Translate (Move/Array) Sketcher function

PossiblyMax, 2025-09-09:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I am new to AstoCad (and FreeCad, and pretty much any Cad) so I might be misunderstanding this.

When using the `Move/Array Transform` button in Sketcher with a group selected, the individual elements are copied along with their constraints, but the group constraint is not replicated to the new copies.

I have tried to illustrate the issue in this short video:

https://github.com/user-attachments/assets/c9458178-0a65-4d50-8f5a-0d9f272b0c11

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.43302 (Git)
Build date: 2025/09/07 16:04:12
Build type: Release
Branch: AstoCAD
Hash: b59392fac2524247fe0038154fca291300d85b68
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/93.2319/1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

PaddleStroke, 2025-09-09:
You are not misunderstanding. It should work and it doesn't. The new constraint is not considered by the array tools (polar and scale also probably don't handle it)

PaddleStroke, 2026-11-16:
Will be fixed in next build (in a few days)